xarial/codestack #5: SOLIDWORKS- CREATE DXFS OF ALL SHEETMETAL PARTS IN OPENED ASSY

ALANApplesauce, 2020-09-04:
I was hoping i could get some help with a macro creation. I need a macro that will create all dxfs of sheetmetal parts in an open assembly. The dxf needs to have the following entities (geometry, hidden edges, library features, forming tools) . I have found other macros online that does a similar task but not exactly what i need. Hope you guys can help!

artem1t, 2020-09-08:
Hi Alan,

Thank you for the submission. We will add this macro to the library. I think it will benefit others in the community as well.

Thanks,
Artem

ALANApplesauce, 2020-09-14:
Thanks!  When will this macro be available?

artem1t, 2020-09-15:
I will try to push it this week

artem1t, 2020-09-19:
Macro has been published: [Export flat patterns from SOLIDWORKS assembly components](https://www.codestack.net/solidworks-api/document/sheet-metal/export-assembly-components/)

closed 2020-09-19 by artem1t

ALANApplesauce, 2020-09-24:
Hi! Thanks for your help with this I really appreciate it! I was wondering if there was a way to prevent the "save modified documents"

artem1t, 2020-09-24:
Hi Alan,

Is this dialog box comes at the end of the macro or it blocks the macro execution? For me, it is only happening in the end. I also not quite like it, but this is unfortunately this particular API behavior, i.e. exporting flat patterns behind the scenes invokes suppressing-unsuppressing which marks the document as dirty. I also prefer not to call save from the macro - just in case. But for now I have created separate [bug](https://github.com/xarial/codestack/issues/13) for this.

Thanks,
Artem

ALANApplesauce, 2020-09-24:
>
> Artem,

I am able to execute the macro and it does effectively create the Dxfs as
requested . The “save all modified documents “ dialog box I am referring to
occurs during the execution of the macro , which requires a response for
the macro to continue executing .  Basically I have to constantly click the
save button repeatedly until all Dxfs have been created. Hopefully your
team can find a way to work around this so there isn’t so much manual
intervention. Other than that , the macro works great ! Thanks ! Hope to
hear from you soon.

artem1t, 2020-09-24:
Hi Alan,

Yes, this is not good as it is blocking. Strange that I do not see the same behavior, i.e. I can export 100s of components automatically. Perhaps I have defaulted some SW option so it doesn't ask - I will work on that - definitely a blocker.

ALANApplesauce, 2020-09-24:
Ok sounds good . Hope to hear from you soon. Let me know if there is
anything I can do to help. Thanks

tudonghoacokhi, 2020-09-24:
(Export flat patterns from SOLIDWORKS assembly components) is doing stable. But I hope a macro like this is even more advanced. For example with assembly there are parts part is copy 2nd, copy 3rd. The macro creates the same DXF with the name 2nd, 3rd. It is very effective when doing complex assemblies, especially when nesting, without having to choose a number of each DXF.

dmadsen4, 2021-06-09:
is there a way to incorporate into this macro a way to remove the tangent edges shown in the modle so they do not show up in the DXF?