xarial/codestack #9: Creat flat pattern view with annotation

Brent-bai, 2020-09-13:
Get the BOM of assembly, and create each Sheetmetal part flat pattern drawing in one slddraw file, and get the material, thickness, and quality value of the parts as annotation, list below the view. The part view that with same material and thickness  will list in the same row.

artem1t, 2020-09-15:
Thank you for the submission. Could you please attach sample data? Original assembly and expected result.

Brent-bai, 2020-09-20:
Hello! Thanks for response.
Here is the file link. [https://drive.google.com/file/d/1gpRGS9FJhCQP_lzdj8BVAC15NchtMLgM/view?usp=sharing](url)

Brent-bai, 2020-09-20:
Hello! Thanks for response.
Here is the file link.
https://drive.google.com/file/d/1ttnfZl_V3FZrq_pDWB_FrsQHhcABU7Cc/view?usp=sharing

artem1t, 2020-09-21:
Thank you for the models. I will be investigating this request

Brent-bai, 2020-09-22:
That's OK.

tpharber, 2021-11-12:
H!s this made any advancement? This would awesome! I've got a macro that will use the bom to spam out all the parts of an assembly on one sheet , but can't get it to convert to flat pattern .