xarial/codestack #12: Solidworks Macro for Balloons in drawing

Johny87v, 2020-09-21:
Hello,

I've been searching the internet everywhere for a macro that will allow me to check for missing balloons and possibly add them to a drawing in solidworks.

I deal with assemblies with 500+ parts at times so its a very painful process at the moment.

thanks
John

artem1t, 2020-09-21:
Cool idea! I will be looking into this. The main trick here is proper positioning of the balloon so I might just skip this step. Instead, macro will scan the view, find all missing balloons, add ones into default location and notify which ones were added for the user to manually align them. Do you think this would be a viable solution?

Johny87v, 2020-09-21:
Yes I think that solution would be an awesome starting point.