xarial/codestack #13: Dirty documents after running the Export Flat patterns macro

artem1t, 2020-09-24:


artem1t, 2020-09-24:
As reported this message blocks the execution so needs to be resolved

ALANApplesauce, 2020-09-25:
Artem,

I wanted to follow up on this......

artem1t, 2020-09-28:
Hi Alan,

I was not able to reproduce the issue. But there were quite a few revisions of the macro. Could you please try the latest version from the site? If the issue is still reproducible, would it be possible to have a remote session with you to troubleshoot? If so, please send me an e-mail to info@xarial.com and we will organize the session.

Thanks,
Artem

ALANApplesauce, 2020-09-28:
Let me try and run the new version first. Do I just go to the original post and copy the code from there? Or where is the updated code located at?

artem1t, 2020-09-28:
> Let me try and run the new version first. Do I just go to the original post and copy the code from there? Or where is the updated code located at? 

Yes, macro is at original post

ALANApplesauce, 2020-09-28:
ok thanks!!

ALANApplesauce, 2020-09-29:
Looks like the updated version works! Thanks for your help! I can't thank
you enough!

artem1t, 2020-09-29:
Great, I will close the ticket. Could you please attach the original macro here if you still have it (the one which doesn't work). As I have about 5 revisions of this macro - I want to try to isolate which one is exactly caused the problem so will not use it in future.

closed 2020-09-29 by artem1t

ALANApplesauce, 2020-09-30:
Attached is the macro.