xarial/codestack #16: Add placeholder for quantity for 'Export Flat Patterns from Assembly'

artem1t, 2020-09-24:
Add <_Qty_> placeholder to be resolved to the quantity of the specified flat pattern. This should represent the flat quantity of the component in the assembly but also the quantity of the identical sheet metal bodies in the part

tudonghoacokhi, 2020-09-24:
> Add <_Qty_> placeholder to be resolved to the quantity of the specified flat pattern. This should represent the flat quantity of the component in the assembly but also the quantity of the identical sheet metal bodies in the part

I hope it automatically chooses the same number of parts in its assembly. Renamed by number, it is possible that many identical DXFs. They are copied from one part, there are many identical parts in the assembly, but actually there is only one part file

artem1t, 2020-09-24:
Do you mean component instances? If you can attach an example file and expected result - this would be great.

Designer-Duane, 2021-12-13:
This Macro does a fine job of giving the Quantity of the flat patterns required for a individual part using:
 Const OUT_NAME_TEMPLATE As String = "<$CLPRP:Sheet Metal Thickness> <_FileName_> <$CLPRP:QUANTITY> <_FeatureName_> .dxf"    

Is there away to get this macro to do the same when running it on an assembly where there a multiples of the same part in the assembly?

Any help would be greatly appreciated.