xarial/codestack #18: A way to name next component number in assemby

france-tarantino, 2020-10-08:
Hi everyone!
First of all I wanted to thank you for the work you are doing, very useful for the community that wants to improve themselves with solidworks.
In this regard, I wanted to ask you if there is a way with "VBA" to solve my following problem:
When I create a new assembly, and insert new virtual components (parts) inside it, it would be very helpful if there was a macro that automatically assigns a progressive numbering to their name. It would be perfect if this macro could be applied instantly as the new part (virtual component) is inserted.
Do you think it is possible? Thank you in advance for your kindness!

artem1t, 2020-10-08:
Thank you for your request. That sounds like a pretty good idea. I have put it into 'Under Investigation' milestone, but I think API would allow implementing this. The only thing it might be better to be implemented as an add-in rather than macro as it needs to constantly monitor the activity. So it might be a good candidate for [CAD+ Toolset](https://cadplus.xarial.com/). Either way we will try to add this to a library

france-tarantino, 2020-10-09:
Thank you @artem1t !
I was thinking that, in the GOODIES / MODELS section you had added a solution to automatically generate the newly created part number by drawing the numbering from a ".txt" file. Something similar could be considered, but which instead allows the activation of the macro once you have entered the virtual components ....
Anyway, I stay tuned to any of your updates ... it would be very useful! Thanks again for your availability and competence!