xarial/codestack #20: Custom properties for 'Export Flat Patterns from Assembly'

Snake-60, 2020-10-22:
Need to be macro take custom properties from active configuration of the part, not only from cut list properties.

Andrius01, 2020-11-27:
Hello,

this macro is working great, I was looking for it for a long time! Thank you for that!

But talking about the perfection I also need to get the exported file name associated with "Custom Properties" values. For example I'm using other macro which counts all the parts in assembly and writes that value in the "Custom Properties" with the name of "Quantity". At this point I'm writing quantity of different parts by hand and it takes a long time... 

Making this process automated would really help me out. 

P.s. Special program https://centralinnovation.com/solidworks/extensions/task-v2/ had this type of export and I was using this program until It became paid.

closed 2020-12-13 by artem1t

Andrius01, 2020-12-16:
Wow... Amazing work! That's exactly what I needed 👍