xarial/codestack #24: Macro Request: Excel File Part Number and Description Manager

tfinley, 2020-11-15:
Per your comment on reddit, I am submitting a macro request.  The purpose of this macro is to keep an up to date excel file of all part numbers and their descriptions.  The idea is to run this macro either on the initial save or as a save as after a concept part is worth of a dedicated part number.  Let's start with the initial save idea:

1. User clicks macro button to run macro
2. Macro opens hard coded excel file (excel file has two columns already titled Part Number and Description with a few entries)
3. Macro identifies the last part number in the first column (eg 123456)
4. Macro saves part to default location (or asks user for location) with file name as the next part number increment - 123457
5. Macro enters 123457 in the first column of the next row (under 123456).
6. User is prompted for a description and user enters the text (e.g. Left Gusset)
7. Macro enters this descriptions in the second column next to the part number 123457.
8. Macro enters description into part description of Solidworks part. (same description that shows up in the BOM table).

Optional: Description Update

1.  User clicks macro button to run macro
2. Macro opens excel file
3. Macro sees that the filename already has an entry.
4. Popup asks user if they would like a new part number or if they want to update description
5. user selects update description
6. User enters new description into text window and selects OK
7. Macro overrides the description next to the part number (wherever it was - not in a new row)

artem1t, 2020-11-15:
Thank you. Just thinking it might be better to use a database, e.g. SQL express or SQLite just to properly deal with concurrency as there might be multiple users and excel file might be locked. Or if the concurrency is not a problem, CSV would be easier than Excel as there is no need to deal with Excel API (or Excel reader) so performance would be better. If CSV can be used it is a good candidate for a macro, for DB or Excel it is better to add this to [CAD+](https://cadplus.xarial.com/) as I do plan to have this model there anyway.

tfinley, 2020-11-15:
CSV is equally useful in my opinion.  DB is a little unfamiliar so not preferred.

I'll look into CAD+