xarial/codestack #25: Auto create the points

Brent-bai, 2020-11-21:
There are many holes in part A, and part B also need to create holes align the center of holes with part A.
I think  the "CylinderParams" property may resolve this problem. I try it with macro in part, it is worked, but not in assembly.
Do you have any ideas?

artem1t, 2020-12-08:
As far as I understand the issue is an incorrect coordinate in the assembly space. Am I right? If so you need to convert coordinates using the transformations. Check [Assembly component transformation](https://www.codestack.net/solidworks-api/geometry/transformation/assembly-component-transformation/) article

Brent-bai, 2021-04-30:
> As far as I understand the issue is an incorrect coordinate in the assembly space. Am I right? If so you need to convert coordinates using the transformations. Check [Assembly component [...]
### ### 
Thanks! It worked. 
Another question:
 How to constrain the points to the hole center?

Brent-bai, 2021-05-05:
I have resolved the points constrain issue. When the points created, it's auto-selected, all I need to do is selecting one edge of the cylinder which is a circle, and add the constrain.