xarial/codestack #27: Macro Request: Export all underlying drawings from Assembly to PDF

Quasardian, 2020-11-26:
We're looking to digitizing our printed SolidWorks drawings to PDF's.
For this we need a Macro, which to my knowledge does not exist yet.

I've seen multiple PDF export macro's but none of which would fit our needs.

We have many different drawings in underlying parts and assembly's in a main assembly, which we would like to export all at the same time.
But since this different for each project we would also need to be able to select specified sheets and a folder.
I put an example in the attachment, which I made with Excel and Paint.
I tried to optimize the layout as best I could, but I'm open to suggestions.

I hope someone can help me with this, thank you in advance!

[image]

Quasardian, 2020-12-15:
I found a macro which opens every underlying file from an assembly and exports the drawing to a PDF, but without any customization.
It might help.?

[https://forum.solidworks.com/servlet/JiveServlet/download/115350-1-156849/Copy_Export+from+assy+to+PDF+DXF+BOM.swp.zip](url)

artem1t, 2020-12-16:
I am planning to add this feature to CAD+ toolset

Quasardian, 2020-12-16:
Thank you!
Is there anything I should keep in mind?
If it will take a long time to develop, or that this will be added to a premium version in the future?

Quasardian, 2021-02-01:
Moved to CAD+

closed 2021-02-01 by Quasardian