xarial/codestack #29: Macro for sheet metal

kaushalsuthar, 2020-12-07:
Can any one help for making macro on making auto dxf ?
dxf should be save like below folder sequence and at the end we found dxf file.
DXF --> MS --> 2 MM --> part1.dxf

artem1t, 2020-12-08:
Take a look at [this macro](https://www.codestack.net/solidworks-api/document/sheet-metal/export-assembly-components/). It has very flexible naming options.

GABRIEL-GDSCCO, 2021-02-24:
olá eu estou tentando usar essa macro que voce cita mas nao estou conseguindo realizar a troca como foi feita no vídeo de colocar em uma pasta contendo as subpastas com a espessura da chapa como foi feito.

segue imagens para serem analisadas.

[image]
[image]
[image]

espero que possa ajudar.

artem1t, 2021-03-01:
> olá eu estou tentando usar essa macro que voce cita mas nao estou conseguindo realizar a troca como foi feita no vídeo de colocar em uma pasta contendo as subpastas com a espessura da chapa como foi feito.
> 
> segue imagens para serem analisadas.
> 
> [image]
> [image]
> [image]
> 
> espero que possa ajudar.

This should be either:

<$CLPRP:Espressura...> or <$PRP:Espressura...> or <$ASSMPRP:Espressura...>

The <Espressura...> is obsolete