xarial/codestack #30: Create STEP from file explorer

latapieA, 2020-12-07:
Hello everyone,
First of all thank you for your precious work and for the development of SolidWorks add.

As is already existing for open in "quick view" mode an assembly or a drawing : 
https://www.codestack.net/solidworks-api/getting-started/scripts/vbscript/open-quick-view/

I would like to know if it is possible to do the same with the save as step file command ?
The step will have the same name of the sldprt or sldasm, and save in the same directory.

I tried to modify the existing vbScript without result...
Thank you in advance for your support. 

Adrien

artem1t, 2020-12-08:
Thank you for the feedback. Could you please clarify the request a little bit. Do you want to create 'Save As Step' command in the context menu in Windows explorer and when you click this command it creates step next to the source file? Just want to make sure I understand the request.

latapieA, 2020-12-08:
Yes it is exactly that.