xarial/codestack #60: Macro to dissolve all subassemblies in a drawing BOM

mbarbosaalp, 2021-06-04:
Hello, first of all let me congratulate you on your work.

Wouldn't it be nice to have a macro to dissolve all the subassemblies on a bom? This is (probably) the only way to get a cut-list from an assembly in SOLIDWORKS and it is a pain to have to do it manually.

Maybe you could give it some thought.

Thanks.
Manuel Barbosa

Berglid, 2021-06-11:
Hi, 

I don't think you need a macro for what you are describing if it is mainly for use in drawing BOM. 

I think [this video](https://www.youtube.com/watch?v=nGc0oQzXItQ&t=10s) will cover what you need to get a flat BOM structure without having to dissolve the assembly itself. 

You can select parts only to get all the individual parts in the BOM. 

A bonus tip if you have both assembly and weldment is to use Indented, flat numbering and hiding the weldment (so only sub components of the weldment show in the bom)

[image]

mbarbosaalp, 2021-06-15:
Hello,
First of all, let me thank you for taking the time to adress my request.
I don't think I presented my problem clearly. I do not need the parts but the cut list items, and the process of dissolving (or hiding) the parts and assemblies is very time consuming and a bit frustating. I am attaching a video with a very small example of the process, sometimes I have lists with multiple pages and those are a pain.
I figure many people will have this issue since this is the only way (I see) to get a cut list from a whole assembly structure, otherwise I would have do model everything in a single part to take advantage from a cut list and that is not feasable.

I am also attaching a macro that I scrapped together, since my programing skills are practically non existant, that for some reason does the job but leaves the system in an unstable state (SOLIDWORKS actually warns this).

Hope you have the time to take a look. Anyway, thank you very much for your time!

Regards,
Manuel Barbosa

Berglid, 2021-06-15:
Hi, 

Okay,  I think this still might be manageable without a macro but will have to review your attachments which I can't seem to see. Did you reply with an email? I think you have to add the attachment on GitHub to enable us to see them. 

If it requires a macro, I will not be able to help as I'm just starting learning SW API/Macro, and are not part of the contributors to Codestack. Just happened to see your issue as I was leaving my own request.