xarial/codestack #114: Macro SloidWorks

Revolution84, 2021-11-24:
Bonjour,

Je souhaite créer une macro avec bouton de raccourci pour générer facilement un fichier STEP des pièces et assemblages.
Je n'arrive pas à ce que ma macro prenne le nom du fichier en cours pour nommer le STEP

J'arrive à créer la macro avec le mode automatique du module macro de SolidWorks mais je ne maitrise pas encore le langage VBA pour mettre ce paramètre dedans.

Est-ce que quelqu'un à la solution et peut m'expliquer ?

sammadox, 2021-11-27:
Bonjour @Revolution84, est ce que tu peux partager le code du module qui génère le STEP file de Solidworks?

Revolution84, 2021-12-03:
Bonjour sammadox
 voici le code généré par SolidWorks avec l'outil macro interne:
'
******************************************************************************
' C:\Users\Etude\AppData\Local\Temp\swx13308\Macro1.swb - macro recorded on
12/03/21 by etude
'
******************************************************************************
Dim swApp As Object

Dim Part As Object
Dim boolstatus As Boolean
Dim longstatus As Long, longwarnings As Long

Sub main()

Set swApp = Application.SldWorks

Set Part = swApp.ActiveDoc

' Save As
longstatus = Part.SaveAs3("C:\PDM_MOBILEV\TEST\Pièce1.STEP", 0, 2)
End Sub

Le tout fonctionne très bien mais j'aimerais que le nom du step soit celui
du fichier en cours d'execution à l'écran dans solidworks et je ne sais pas
faire.

Merci pour ton aide