xarial/codestack #122: Export flat patterns from SOLIDWORKS part or assembly components

Lookieuk, 2022-01-05:
Hi there am getting this error 

when using the Export flat patterns from SOLIDWORKS part or assembly components 
have changed the code to allow for the output to sort in thickness, cant seem to get it to work

[image]
[image]

all help will be grateful

thanks lookieuk

Skylos32, 2022-01-05:
It's been improved since the video was uploaded. For cut-list custom property you need to specify <$CLPRP:Sheet Metal Thickness>
Check the description for information
https://www.codestack.net/solidworks-api/document/sheet-metal/export-all-flat-patterns/

Lookieuk, 2022-01-05:
thanks very much :) is there a way to change all the parts in my assembly to have the cultist properties added

and not sure if there a way for the qty off each part or same parts within the assembly to show in the file name or dwg drawing 

[image]

Lookieuk, 2022-01-05:
Have another problem the code works on parts only when opened but not on the assembly file, come up with this error

[image]

The parts have cut-list