xarial/codestack #155: Solidworks cutlist rename

bfdesigns, 2022-06-27:
Hope someone can help me as I'm looking for this for years now.
We rename the folders under the Cut List by slowly double clicking it and renaming it manually.
I'm looking for a macro to pull this name trough to the part and give it a part number at the end of this part name that pulled trough.
For example we would name the folder as seen in attached image as GFE-MIR-POR-STR-I BEAM 1
The next folder that is a I BEAM would be GFE-MIR-POR-STR-I BEAM 1
So on for plates, channels, angle irons ect.
In our case the number after the "." indicates that specific part's part number. So like the I BEAM folder shown in this image would range from GFE-MIR-POR-STR-I BEAM 1.1 to GFE-MIR-POR-STR-I BEAM 1.28 as there is 28 identical parts in that folder.
From here we select save bodies to end up with single parts.

If someone could help or point me in the direction of where to find a macro just to pull the custom renamed folder name trough to the part and somehow add an indication of a part number in to the name it would be highly appreciated.
[image]

Blenky56, 2022-11-19:
I have a macro that uses the part file name and converts the name of each item in the cutlist to "filename-01"; "filename-02"; "filename-03"; etc.  It most likely can be modified to pull the folder name from the cutlist and use that instead.  It also populates the cutlist properties for me as well but you are only looking to change the name so that portion of the macro can be removed.

 Let me know if you are interested in it.  

Blenky56

thalysson98, 2023-05-18:
What you can do is you create the folder with the name that you want, and then move the items according to each folder. After that you can run a macro to rename the bodies according to the folder name, for example: GFE-MIR-POR-STR-I BEAM 1.1, GFE-MIR-POR-STR-I BEAM 1.2.