xarial/codestack #170: Create Selectable Bounding Box

OmerF4, 2022-09-02:
Hi,
we would like to make Create Selectable Bounding Box macro parametrically. We mean we created a solid part and we used this macro then we changed the dimension of the part but the size of the box is not changing according to the new solid part. we request from you how can we make it automatically

link : https://www.codestack.net/solidworks-api/geometry/create-selectable-bounding-box/

artem1t, 2022-11-03:
Try rebuilding the model, it should remain parametric

OmerF4, 2022-11-03:
We tried the code in the website again. This time code worked , so we thank
you for it. But there is a thing that we can not get it about it. We tried
the code previously ,  the code did not work although we rebuilt model but
now code is working. I wonder you changed the code. whatever, thank you for
everything

şunu yazdı:

artem1t, 2022-11-03:
The code was not changed (the latest commit was in 2020):

[image]