xarial/codestack #239: Object Doesn't support this property or method

Azrofive, 2023-04-28:


Azrofive, 2023-04-28:
Sorry for hitting enter like an idiot, i am trying to run your script but keep getting the error "Object doesn't support this property or method"
I am running solidworks 2013 and am wondering if this is a problem as the code may of been written for a later version of solidworks? Is this something you have dealt with before and have a solution for or not?
Kind regards!
Jack Tunks
[image]

artem1t, 2023-05-01:
What macro do you run?

Azrofive, 2023-05-01:
Hi Artem,
I was running the assembly to dxf flat pattern macro
Regards
Jack