xarial/codestack #307: Excel macro to manage custom properties in SOLIDWORKS files

OmerF4, 2024-02-28:
Hi,

This macro is really nice. However, I want to ask something about this macro. I wonder how I can read the valueted value column instead of value/text expression when I use the getswprp command in Excel. I would be very grateful if you could help me with this. Because I read the formula itself instead of the result values of the formulas we wrote into the program.

andrMollo, 2024-02-29:
Hi,
not an expert here, but I think you could replace `GetCustomProperty` with `GetCustomPropertyValues`. So:
```vba
Dim prpType As SwDmCustomInfoType
    
    If confName = "" Then
        For i = 0 To UBound(vNames)
            res(i) = swDmDoc.GetCustomProperty(CStr(vNames(i)), prpType)
        Next
    Else
        Dim swDmConf As SwDocumentMgr.SwDMConfiguration10
        Set swDmConf = swDmDoc.ConfigurationManager.GetConfigurationByName(confName)
        If Not swDmConf Is Nothing Then
            For i = 0 To UBound(vNames)
                res(i) = swDmConf.GetCustomProperty(CStr(vNames(i)), prpType)
            Next
        Else
            Err.Raise vbError, "", "Impossibile ottenere la configurazione '" & confName & "' da '" & fileName & "'"
        End If
    End If
```

will become
```vba
Dim prpType As SwDmCustomInfoType
    Dim linkedTo As String
    
    If confName = "" Then
        For i = 0 To UBound(vNames)
            res(i) = swDmDoc.GetCustomPropertyValues(CStr(vNames(i)), prpType, linkedTo)
        Next
    Else
        Dim swDmConf As SwDocumentMgr.SwDMConfiguration10
        Set swDmConf = swDmDoc.ConfigurationManager.GetConfigurationByName(confName)
        If Not swDmConf Is Nothing Then
            For i = 0 To UBound(vNames)
                res(i) = swDmConf.GetCustomPropertyValues(CStr(vNames(i)), prpType, linkedTo)
            Next
        Else
            Err.Raise vbError, "", "Impossibile ottenere la configurazione '" & confName & "' da '" & fileName & "'"
        End If
    End If
```

OmerF4, 2024-02-29:
> Hi, not an expert here, but I think you could replace `GetCustomProperty` with `GetCustomPropertyValues`. So:
>
> ```vba
> Dim prpType As SwDmCustomInfoType
>
>     If confName = "" Then
>         For i = [...]

OmerF4, 2024-02-29:
> Hi, not an expert here, but I think you could replace `GetCustomProperty` with `GetCustomPropertyValues`. So:
> ```vba
> Dim prpType As SwDmCustomInfoType
>
>     If confName = "" Then
>         For i = [...]

Hi AndrMollo,
I am so grateful for your help. the code that you wrote above worked.

closed 2024-02-29 by OmerF4